neskweek/LightSaberOS #65: Erreur compilation Arduino/ Uno

Tam-Koowan, 2017-07-01:
J'essai de téléverser sur une CArte Adruino Uno mais j'ai cette erreur :
C:\Users\Chris\Documents\Arduino\libraries\LightSaberOS-LSOS-1.5_GravityColorBlend\LightSaberOS\LightSaberOS\LightSaberOS\LightSaberOS.ino:22:24: fatal error: DFPlayer.h: No such file or directory

   #include <DFPlayer.h>

                        ^

compilation terminated.

exit status 1
Erreur de compilation pour la carte Arduino/Genuino Uno

Pouvez-vous m'aider  ?

Merci

Tam-Koowan, 2017-07-04:
J'ai suivi le tuto à la lettre mais rien à faire !!!

Tam-Koowan, 2017-07-09:
Oui j'ai mis le dossier Librarie dans mon Dossier Arduino/librairie...
Mais ça marche pas  :(

Tam-Koowan, 2017-07-16:
After several modifications I have no blatant mistake of the blow I do not understand the problem ...
Can you help me ?
I have an arduino Nano V3 for this achievement.
Here is the error report:

Arduino : 1.6.11 (Windows 7), Carte : "Arduino Nano, ATmega328"

In file included from C:\Program Files (x86)\Arduino\libraries\LightSaberOS\LightSaberOS.ino:37:0:

sketch\Config.h:263:9: warning: extra tokens at end of #else directive

   #else if DIYINO_STARDUST

         ^

sketch\Config.h:274:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST 

       ^

sketch\Config.h:332:7: warning: extra tokens at end of #else directive

 #else if DIYINO_STARDUST

       ^

C:\Program Files (x86)\Arduino\libraries\LightSaberOS\LightSaberOS.ino:1035:13: warning: extra tokens at end of #else directive

       #else if STAR_LED

             ^

In file included from C:\Program Files (x86)\Arduino\libraries\LightSaberOS\LightSaberOS.ino:33:0:

C:\Program Files (x86)\Arduino\libraries\FX-SaberOS/LinkedList.h: In instantiation of 'LinkedList<T>::LinkedList() [with T = unsigned int]':

sketch\SoundFont.h:30:14:   required from here

C:\Program Files (x86)\Arduino\libraries\FX-SaberOS/LinkedList.h:101:6: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]

  root=false;

      ^

C:\Program Files (x86)\Arduino\libraries\FX-SaberOS/LinkedList.h:102:6: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]

  last=false;

      ^

C:\Program Files (x86)\Arduino\libraries\FX-SaberOS/LinkedList.h: In instantiation of 'LinkedList<T>::~LinkedList() [with T = unsigned int]':

sketch\SoundFont.h:30:14:   required from here

C:\Program Files (x86)\Arduino\libraries\FX-SaberOS/LinkedList.h:121:7: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]

  last = false;

       ^

C:\Program Files (x86)\Arduino\libraries\FX-SaberOS/LinkedList.h: In instantiation of 'bool LinkedList<T>::add(T) [with T = unsigned int]':

sketch\SoundFont.h:352:15:   required from here

C:\Program Files (x86)\Arduino\libraries\FX-SaberOS/LinkedList.h:192:12: warning: converting 'false' to pointer type 'ListNode<unsigned int>*' [-Wconversion-null]

  tmp->next = false;

            ^
[... 518 lines cut ...]

C:\Program Files (x86)\Arduino\libraries\LightSaberOS/LightSaberOS.ino:491: undefined reference to `OneButton::attachLongPressStop(void (*)())'

C:\Program Files (x86)\Arduino\libraries\LightSaberOS/LightSaberOS.ino:492: undefined reference to `OneButton::attachDuringLongPress(void (*)())'

C:\Users\Chris\AppData\Local\Temp\ccV5dPyt.ltrans2.ltrans.o: In function `DFPlayer::setSerial(unsigned char, unsigned char, bool) [clone .isra.2]':

C:\Program Files (x86)\Arduino\libraries\LightSaberOS/DFPlayer.h:148: undefined reference to `DFPlayerSerial::DFPlayerSerial(unsigned char, unsigned char, bool)'

C:\Program Files (x86)\Arduino\libraries\LightSaberOS/DFPlayer.h:150: undefined reference to `DFPlayerSerial::begin(long)'

collect2.exe: error: ld returned 1 exit status

exit status 1
Erreur de compilation pour la carte Arduino Nano
Bibliothèque non valide trouvée dans D:\Documents\Arduino\libraries\MMA8451-master : D:\Documents\Arduino\libraries\MMA8451-master
Bibliothèque non valide trouvée dans D:\Documents\Arduino\libraries\Servo_2_accelerometre : D:\Documents\Arduino\libraries\Servo_2_accelerometre
Bibliothèque non valide trouvée dans D:\Documents\Arduino\libraries\_2_Servo_ACC : D:\Documents\Arduino\libraries\_2_Servo_ACC
Bibliothèque non valide trouvée dans D:\Documents\Arduino\libraries\MMA8451-master : D:\Documents\Arduino\libraries\MMA8451-master
Bibliothèque non valide trouvée dans D:\Documents\Arduino\libraries\Servo_2_accelerometre : D:\Documents\Arduino\libraries\Servo_2_accelerometre
Bibliothèque non valide trouvée dans D:\Documents\Arduino\libraries\_2_Servo_ACC : D:\Documents\Arduino\libraries\_2_Servo_ACC

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

azsde, 2017-07-20:
Tu as fait comment pour copier le projet ?

Moi je l'ai cloné depuis github et il la compilation fonctionnait directement.

Tam-Koowan, 2017-07-20:
J'ai télécharger le fichier zip, puis dé zipper et mis dans le dossier arduino !

Tam-Koowan, 2017-07-20:
Malgré avoir cloné le projet ça ne change rien j'ai toujours un problème de compilation  ! :(