neskweek/LightSaberOS #67: Lsos 1.5 gravity and more than 120pixels

faeton13, 2017-07-10:
Hi ! I have to use  more than 120pixels and seems find issue if you choose 120+ pixels gravity function stops working. Blade glows with lime-orange and indication segment on blades top doesn't glow up.

jbkuma, 2017-07-10:
How many pixels are you trying to use?

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

faeton13, 2017-07-10:
132 but that doesn`t matter. it change behavior from 121 choise. 
There was same sort of problem at 1.4 when it turns pixel count to 0 right after 120, so 125 light up just 5 pixels

Protonerd, 2017-07-11:
There is no limitation per se, except that you might run into stability problems due to low dynamic memory available to the other functions. If this is the case, the software will simply freeze. That is what you might see during the color mixing. Can you still operate the saber once it happens?

faeton13, 2017-07-11:
Ok made some testing today.
1 Lower pixel number and calibrate mpu. Colour mixing start working but I can switch only red and blue, and can't find position where's green in colour mixing "sphere". 
2  max pixel number when curent color tip is vizible on  blade tip is 127. After - saber is still working, but  Gravity mixing colour tip seems spread all over blade. I can't find any 128number in code. 
É

Protonerd, 2017-09-26:
Is this topic resolved? Or still need support?