neskweek/LightSaberOS #68: DFPlayer Behavior on LSOS 1.5

azsde, 2017-07-30:
I am having troubles with my DFPlayer, the volume is correctly set, but it keeps looping on "Soundfont" sound and doesn't play anything.

I have tried with the old library, and it is sometimes working.

What can I do to fix this ?

DRagonenko, 2017-07-30:
try the modes with the usual LEDs on newpixel it happens

azsde, 2017-07-30:
I kinda figured it out, why my arduino nano is loaded with too much code (90% of the space), the whole system behaves weirdly...

I think I'll rewrite some features when I have to time so that they are cleaner and lighter

azsde, 2017-08-04:
I'm sorry to re-open this issue but I'm really struggling getting this feature to work.

Can anyone help me ?

azsde, 2017-08-29:
It was an hardware issue, switched to another dfplayer and it works fine

closed 2017-08-29 by azsde