neskweek/LightSaberOS #69: Button behavior is different is powered in 3.7V or 5V

azsde, 2017-07-30:
When running from 5V usb power source, the button is behaving just fine, but when running on a 3.7V battery, a single press is interpreted as a long press.

Is it normal ?

jbkuma, 2017-07-30:
I've only ever used batteries or a 3.7v wall supply and have never seen
this.

azsde, 2017-07-30:
@jbkuma sorry for the inconvenience, it seems indeed that the observed behavior isn't related to the power supply, but rather to the arduino capacity. 

When I load up to 90% or more of its program memory, it behaves very weirdly.

Maybe it is due to the fact that the arduino is a clone and not a genuine one.

azsde, 2017-08-04:
This issue can be closed, I figured out that the ticks had to be adapted.

closed 2017-08-04 by azsde