neskweek/LightSaberOS #70: sound font problem？

bigjonyz, 2017-09-16:
Hi there, great work for aspiring saber builders. I have just started building my saber. I had worked with jakesoft's usaber before, now I am trying yours. I made the hw modifications and used a simple led on pin4. The problem is when i boot up i hear a click On the speaker and see the serial print as dfplayer connected, but no sound is played. I tried to set soundfontQuantity to 1 or 2 no luck. Reformat and drag and drop files no luck.  Upload the dfplayer basic test program all files are played, serial print indicate they are in the right order. soundfont.h file id numbers i have not changed and using the supplied sdcard files with no changes to names or order. What Could be the issue here? Also pressing the main button the led will lightup, so the rest of the code should be fine, just the sound.

kirkyb, 2017-09-16:
I'm experiencing the same issue. I hear the "Diyino" boot sound and then everything is jumbled after that

hummeruli, 2017-09-26:
For me also the same trouble

Protonerd, 2017-09-26:
This is among those phenomena I never got around to find the root cause for, so I had to settle with workarounds. The things is, sometimes, and it does not happen often, if you change any part of the code, not in any how related to the DFPlayer, there is no compilation error or such, but sound simply cases to be played. Only workaround is to reshuffle the same code, or insert some dummy lines, then it starts working again. I guess this is a compiler issue and can happen any time the code is changed at any inconspicuous place. Anyway, as LSOS slowly gets obsolete, I would suggest you guys to move to FX-SaberOS. As to the problem mentioned by kirkyb, I think this might be different since at least the intro is played. In this case I would suggest to have a look at the serial monitor to see exactly where the code stucks.

kirkyb, 2017-09-26:
I have given up with the nano and decided to purchase a Stadust V2 since the code will be optimised for this board and i will be supporting its development :-)

hummeruli, 2017-09-28:
@Protonerd    Thanks, i moved to FX-SaberOS and it works perfect.

Himmiko, 2017-11-15:
I have installed FX-SaberOS,but the problem remained((