neskweek/LightSaberOS #71: battery

DRagonenko, 2017-10-01:
how to make a full charge the battery was at 4.2 volts and not 5 ?

jbkuma, 2017-10-01:
It's you are using a lithium battery the full is 4.2. If you are using a
boost circuit you need to read the actual battery.

DRagonenko, 2017-10-01:
What needs to change in the code, so it is with the battery read?

jbkuma, 2017-10-01:
Are you trying to read a 5v source? You can't accurately read it, changing
the settings won't fix that. The booster draws more current to boost the
voltage as the battery discharges. You need read the battery directly.

DRagonenko, 2017-10-01:
I want to read the powered with 18650 battery maximum 4.2 minimum 3.3

DRagonenko, 2017-10-01:
[image]
at random understood