neskweek/LightSaberOS #73: Max brightness and D8-D11

Phobos06, 2018-01-28:
Hi all
this project is superb!

I'm tryng to do my lighsaber with the LSOS version 1.5.

My configuration is a Neopixel led string "single blade" (and DFPLAYER + MPU6050)

Everything works well but now i'm trying to understand how MAXBrightness works in the saber.
Can anyone explain it to me?

Another question.
In my brain, the 2 button(Main+Config) have a led.
If you are in CONFIG MODE, LED1 is powereod otherwise, if the lightsaber is on, the LED2 is Poweredon.

I've tryed to connect two led to ports D8...D11 (I've check more combination) and in the button class, where there are the "lighsaber poweron/poweroff" and "enter/exit configmenu" i add the "digitalWrite". i've changed also the value of LS1...LS6 in the class config (set all to 3)

But the STATUS LED are in anarchic status :) or better, they switch-on when the saber poweron.
(probably there is "something" that interfes.. but what?)

Any Help?

Thanks a lot
Have a nice day!
Alan

jbkuma, 2018-01-28:
MAX_BRIGHTNESS is pretty straight forward.. it's the maximum brightness.
If it's 200, the brightness won't exceed 200 (out of 255).

If you are trying to code your own button accent mods, you can code your
accents to follow the state mode:
enum SaberStateEnum {S_STANDBY, S_SABERON, S_CONFIG, S_SLEEP, S_JUKEBOX};

Following the action states isn't currently quite as straight forward, you
can try to follow the action states as well as some of the flags like
lockuponclash.

I have my own heavily modified version of the code, so I'm not really sure
exactly how the current public version of the code works in this regard,
but I believe the current sequence is that accent should be lit when the
blade is lit and flash when in lockup on clash or blaster block mode.

Jason "Kuma" Brinkerhoff
*Mad Science Workshoppe, *proprietor
http://jbkuma.com/workshoppe

Phobos06, 2018-01-28:
Thanks for your fast reply to my post!

my question about MAX_BRIGHTNESS  was more basic :)
how it works? 

- on software side? (so, the PWM data signal connected to my neopixel ledstring?) (in this case, i change the parameter parameters and i've finished)
- on hardware side? for example there is an analogic output to connect to some Transistor? or something else? 

_____

About the second question, i'm wondering what D8-D11 are configured to do (and where) in order to force them to change the output when i want (and not when they want) :)

thanks again

Alan

jbkuma, 2018-01-29:
You seem to be way over thinking it.

The mode of operation is different for the different blade types. My best
advice is to search for the term and read the functions.

Phobos06, 2018-01-29:
OK, i will try.

Yesterday evening i solved the STATUS LED problem.

Now i wil work on MAX_BRIGHTNESS side :)

Thanks again